Frankkkkk/python-pylontech #9: module 'pylontech' has no attribute 'Pylontech'

VictorBucha, 2021-12-02:
Hi, @Frankkkkk 
this is what I've got from the very beginning...

root@raspberrypi:~# pip3 install python-pylontech
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting python-pylontech
  Using cached https://files.pythonhosted.org/packages/19/29/caedde3663ae093c2046ed70925b4f2cc673817f7255d7b2ae9e4dc92e63/python_pylontech-0.0.6-py3-none-any.whl
Requirement already satisfied: pyserial in /usr/lib/python3/dist-packages (from python-pylontech) (3.4)
Requirement already satisfied: construct in /usr/local/lib/python3.7/dist-packages (from python-pylontech) (2.10.67)
Installing collected packages: python-pylontech
Successfully installed python-pylontech-0.0.6
root@raspberrypi:~# python3
Python 3.7.3 (default, Jan 22 2021, 20:04:44)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pylontech
>>> p = pylontech.Pylontech()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'pylontech' has no attribute 'Pylontech'
p = import pylontech
>>> p = pylontech.Pylontech()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'pylontech' has no attribute 'Pylontech'

VictorBucha, 2021-12-02:
Don't know why it appeared here as crossed

Frankkkkk, 2021-12-05:
Hi, Sorry for the delay.

You were right, the package didn't import the module, so you had to `from pylontech.pylontech import Pylontech`. I've fixed it to conform to the doc.

I've released 0.0.7 on pypi. It should now work if you update the package.

Thanks for the report !
Cheres

closed 2021-12-05 by Frankkkkk