Frankkkkk/python-pylontech #10: Only get_system_parameters() working

JumpmanJunior, 2022-01-13:
I am trying to use the library with a Pylontech 2000C. But only the get_system_parameters() command is working. I get no reply to all the other commands. Here is the error message - I am also printing the raw frame which I receive from the Pylontech:
```
>>> print(p.get_system_parameters())
b'~20024600B032110E420BEA0AF00D030A470384D2F0B3B0A9EC0D030A47FC7CF272\r'
Container: 
    CellHighVoltageLimit = 3.65
    CellLowVoltageLimit = 3.05
    CellUnderVoltageLimit = 2.8
    ChargeHighTemperatureLimit = 33.31
    ChargeLowTemperatureLimit = 26.31
    ChargeCurrentLimit = 9.0
    ModuleHighVoltageLimit = 54.0
    ModuleLowVoltageLimit = 46.0
    ModuleUnderVoltageLimit = 43.5
    DischargeHighTemperatureLimit = 33.31
    DischargeLowTemperatureLimit = 26.31
    DischargeCurrentLimit = -9.0
>>> print(p.get_values())
b''
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/python-pylontech/pylontech/pylontech.py", line 212, in get_values
    f = self.read_frame()
  File "/home/pi/python-pylontech/pylontech/pylontech.py", line 171, in read_frame
    f = self._decode_hw_frame(raw_frame=raw_frame)
  File "/home/pi/python-pylontech/pylontech/pylontech.py", line 151, in _decode_hw_frame
    assert got_frame_checksum == int(frame_chksum, 16)
ValueError: invalid literal for int() with base 16: b''
```
Any idea what the issue could be?

hnxfirefly, 2022-01-16:
I'm having the same Problem but got it fixed with selecting the ID of the Battery Back one by one
```
    def get_values_single(self, dev_id):
        bdevid = "{:02X}".format(dev_id).encode()
        self.send_cmd(dev_id, 0x42, bdevid)
        f = self.read_frame()
        # infoflag = f.info[0]
        d = self.get_values_single_fmt.parse(f.info[1:])
        return d
```
`    get_values_single_fmt = construct.Struct(
        "NumberOfModules" / construct.Byte,
        "Module" / construct.Array(1, construct.Struct(
            "NumberOfCells" / construct.Int8ub,
            "CellVoltages" / construct.Array(construct.this.NumberOfCells, ToVolt(construct.Int16sb)),
            "NumberOfTemperatures" / construct.Int8ub,
            "AverageBMSTemperature" / ToCelsius(construct.Int16sb),
            "GroupedCellsTemperatures" / construct.Array(construct.this.NumberOfTemperatures - 1, ToCelsius(construct.Int16sb)),
            "Current" / ToAmp(construct.Int16sb),
            "Voltage" / ToVolt(construct.Int16ub),
            "Power" / construct.Computed(construct.this.Current * construct.this.Voltage),
            "RemainingCapacity" / DivideBy1000(construct.Int16ub),
            "TotalCapacity" / DivideBy1000(construct.Int16ub),
            "CycleNumber" / construct.Int16ub,
        )),
        "TotalPower" / construct.Computed(lambda this: sum([x.Power for x in this.Module])),
        "StateOfCharge" / construct.Computed(lambda this: sum([x.RemainingCapacity for x in this.Module]) / sum([x.TotalCapacity for x in this.Module])),

    )
`

Frankkkkk, 2022-01-16:
I, sorry for the late reply.
@hnxfirefly  would you mind doing an MR so we can fix the code ?

@JumpmanJunior  I wasn't able to reproduce on my battery. I'll check your payload ASAP and try to fix the code if @hnxfirefly doesn't fix it first :p
Cheers

JumpmanJunior, 2022-01-20:
@hnxfirefly solution worked. I was able to get the values with his code and requesting the values for each battery individually.

hnxfirefly, 2022-01-20:
@Frankkkkk I'm planning to do a MR but i haven't had time till now as i'd like to set define the struct in a way it makes sense for a single request and also a function that requests every battery one after another.

eperdeme, 2022-12-13:
Single also works for me on my 3000C's