Frankkkkk/python-pylontech #12: Can’t get any data from US2500

n4dmh, 2022-03-25:
Hi!
I’ve tried all combinations of A/B
And DIP on/off (changing baud rate in code) and using L0/L1
But I don’t seem to get any answers from the battery. Currently the battery is not connected anywhere, but SOC does flash so I suppose it should reply in RS485?

Could this be just because the model is 2500  and therefore it has some different commands? 

or do I have some other issue?

what do you think, Frank?

thanks, 
Dennis

demoze, 2022-05-04:
UP2500 RS485 protocol is completely different from US20000|US3000 version.  it won't work.  I'am also looking for the specification of the protocole. ;-)

abelsson, 2022-09-04:
I have a UP2500 and started playing with this code today and have gotten some basic messages to work - I don't think it's too different from what's already implemented. It's too soon to open a MR, but I have some things pushed to my fork (https://github.com/abelsson/python-pylontech) which allows me to query status of a single battery in a way similar to #10.

demoze, 2022-09-04:
Good job, do you have the specification for the UP2500 ?

abelsson, 2022-09-05:
Afraid not, it was just trial and error.

Sven21279, 2022-10-23:
Moin, (Hello)

I bought a second battery this week. I use RS485 for readout.

you can use "~20024642E00202FD33\R" for the first batterie and "~20034642E00203FD31\R" for the second batterie.

I can readout both of my blocks using pythonscript in node-red.

regards from germany
sorry for my english.