Frankkkkk/python-pylontech #13: Question: How to use different serial port than ttyUSB0?

df8oe, 2022-08-25:
Can I tell pylontech.py to use e.g. ttyUSB11 instead of ttyUSB0? Everything is working fine except my USB devices (I have multiple) are differently enumerated...

Frankkkkk, 2022-08-25:
Hi,
You can specify the serial port in the constructor: https://github.com/Frankkkkk/python-pylontech/blob/master/pylontech/pylontech.py#L112

Cheers

df8oe, 2022-08-25:
Thank you for your reply (and your perfectly working project) - but this breaks pip upgrade/update functionality. After an upgrade via pip changes are lost...

Frankkkkk, 2022-08-25:
I don't understand; how do you use the library ?
In your code, you should have something like:
```python3
p = Pylontech(serial_port='/dev/ttyUSB11')
…
```
And that shouldn't break when updating via pip ‽
Cheers

df8oe, 2022-08-25:
The issue is not at your side - it is at my side. I do not have experiences with python and do not want to poke in pylontech.py. This answer rescues my day - thank you!

closed 2022-08-25 by df8oe