Frankkkkk/python-pylontech #25: Problem using RS485-Ethernet gateway

Jschz99, 2023-01-18:
Hi everyone. My name is Jesús, I'm a junior engineer. First of all, thanks for your contribution with the python-pylontech lib, it is a great help. I'm using it to connect my force L1 Pylontech to my RaspBerry Pi and I have a problem with it. If you had time and could answer me, I would appreciate it.

My case is the next one: I'm using a R-485 to Ethernet adapter (RSETHER). I run the next command: 

**socat pty,link=$HOME/bat2,waitslave tcp:<IP_ADDRESS>:<PORT>**

The problem is that I'm not sure what part of the code I've to change. I've change the next line from the pylontech.py file:

**def init​​__(self, serial​​_port='/dev/ttyUSB0', baudrate=115200):**

And I've changed it into:

**def init​​__(self, serial​​_port='/home/jesus/bat2', baudrate=115200):**

(Thats the rute to my virtual serial port) Then I run the script and I get the next fails:

**Traceback (most recent call last):

File "/home/jesus/Desktop/GitHun/python-pylontech-master/pylontech/PRUEBA.py", line 3, <module>
      print(p.get_values())

File "/home/jesus/Desktop/GitHun/python-pylontech-master/pylontech/pylontech.py", line 269, in get_values
      f = self.read_frame()

File "/home/jesus/Desktop/GitHun/python-pylontech-master/pylontech/pylontech.py", line 199, in read_frame
      f = self._decode_hw_frame(raw_frame=raw_frame)

File "/home/jesus/Desktop/GitHun/python-pylontech-master/pylontech/pylontech.py", line 181, in_decode_hw_frame
      assert got_frame_checksum == int(frame_chksum, 16)
Value: invalid literal for int() with base 16: b''**

PRUEBA is the name of my script. And thats everything. I don't understand the problem it is reporting, I don't know if I have to change anything else from the lib.

Again, I would be very grateful with any answer.

Best Regards

Jschz99, 2023-01-18:
Hi again. I've tried with and RS485-usb adapter and I get the same error report. 

[image]

Frankkkkk, 2023-01-18:
Hi,
did you check out that your wiring is ok ? Did you try to permutate A/B wires ? Cheers

Jschz99, 2023-01-18:
Yes, I've done it and I have the same fail... I've checked that my RaspBerry Pi recognizes the USB adapter and apparently it is okey.

[image]

And the point is that now I have the code exactly like you, without any modification (using de RS485-USB adapter)

Frankkkkk, 2023-01-18:
Could you explain how you have wired your cable and maybe attach a picture of the rj45 connector ? Is it plugged into the rs485 port ? Maybe a picture of the pylontech battery(es) ? Are they powered on ?

Jschz99, 2023-01-18:
Yes. As I've said, I've tried two diferent thing. The first one is the RS485-ethernet gateway. This is my battery bms (connected to two batterie modules):

[image]

From there, I take the RS485 wire. The wire I'm using is the one that brings the battery BMS pack:

[image]

[image]

I take the wire to an adaptor to have the two wires in the air without cutting the original wire:

[image]

Then, I take the two wires into the gateway. Then I connect the gateway to a router. I connect the router to my raspberry and using the IP, the port and the virtual serial point, I run the script.

[image]

The other option I've tried is using the rs485-usb adapter instead of the gateway:

[image]

Then, I connect it directly to the raspberry pi.

Frankkkkk, 2023-01-18:
On the 4th picture, are the yellow/white wires attached to the 7/8 pins of the converter ? Is the rj45 cable straight (same wire color on both ends) ?

Jschz99, 2023-01-19:
Yes and yes :)

Frankkkkk, 2023-01-19:
In that case, I'd either propose you to simplify your setup (i.e. cut the rj45 cable and wire it directly to the usb converter using the last two strands (brown-white/brown). And if not, then hook up an oscilloscope :-)

Jschz99, 2023-01-20:
Hi again. I've managed to solve the problem.  If I run the __command p.get_values_single(2_)_ instead of _p.get_values_  for some reason it returns me all the data without any error. I've run the rest of the commands and all of them works perfecly except _p.get_management_info()_, which returns: raise exception (Dont touch this for now). I don't know the reason for this. 

Anyway, thanks Frank for the attention and for the great work you've done with the script :)

Frankkkkk, 2023-01-23:
Hi @Jschz99 thanks for the follow up. I'll try to add a disclamer on the readme for the future users.

Cheers !