Frankkkkk/python-pylontech #29: Making it work into RS232

tomascrespo, 2023-04-10:
Hi Frakkkkk!

I have read your code and it is great

I would like to make it work through RS232 because my Pylon are using RS485 interface to communicate with my inversor.

As far as I know Pylontech RS485 protocol and RS232 are almost the same, but I am not completly sure. 

My batteries communicate with RS232 with BatteryView (Pylontech software) perfectly, using and RS232 to USB adapter. 

I think the first step is putting the serial port in 1200 bps, because this is the default for Pylontech RS232.

Before I deep more into this I want to know if someone has tried before or your knowledge, if it is possible or not

Lots of thanks

tomascrespo, 2023-04-12:
Finally I get it working. At least most of the times.

p.get_values_single(1) works always
p.get_system_parameters() works always

However p.get_values() only works sometimes. Other times it fails with a weird response, I do not why 😕 

What I did:
Change address from 2 to 1 into get_values() and into get_system_parameters()
Modify constructor to allow communication at 4800 speed rate:
```
    def __init__(self, serial_port='/dev/ttyUSB0', baudrate=1200):
        self.s = serial.Serial(serial_port, baudrate, bytesize=8, parity=serial.PARITY_NONE, stopbits=1, timeout=6)
        self.s.write(bytes([0x7E, 0x32, 0x30, 0x30, 0x31, 0x34, 0x36, 0x39, 0x31, 0x45, 0x30, 0x30, 0x32, 0x30, 0x33, 0x46, 0x44, 0x32, 0x46, 0x0D])) # Sets speed to 4800
        time.sleep(1)
        self.s.close()
        self.s = serial.Serial(serial_port, 4800, bytesize=8, parity=serial.PARITY_NONE, stopbits=1, timeout=2)
        self.s.isOpen()
        self.s.flush()

``` 
Change read_frame() to read until '\n' (end of frame)
```
    def read_frame(self):
        #raw_frame = self.s.readline()
        raw_frame = self.s.read_until(b'\r')
        f = self._decode_hw_frame(raw_frame=raw_frame)
        parsed = self._decode_frame(f)
        return parsed
```
As mentioned, **p.get_system_parameters()** does never fail.
```
----> Sending:  b'~200146470000FDA8\r'
7e323030313436343730303030464441380d
<---- Receiving:  b'~20014600B032000E420BEA0B540D0D0A3D03FCD2F0B3B0ADD40D0D0A3DFC18F240\r'
7e3230303134363030423033323030304534323042454130423534304430443041334430334643443246304233423041444434304430443041334446433138463234300d
Container:
    CellHighVoltageLimit = 3.65
    CellLowVoltageLimit = 3.05
    CellUnderVoltageLimit = 2.9
    ChargeHighTemperatureLimit = 61.0
    ChargeLowTemperatureLimit = -11.0
    ChargeCurrentLimit = 10.2
    ModuleHighVoltageLimit = 54.0
    ModuleLowVoltageLimit = 46.0
    ModuleUnderVoltageLimit = 44.5
    DischargeHighTemperatureLimit = 61.0
    DischargeLowTemperatureLimit = -11.0
    DischargeCurrentLimit = -10.0

```

**p.get_values_single(1)** does not fail:
```
----> Sending:  b'~20014642E00201FD35\r'
7e3230303134363432453030323031464433350d
<---- Receiving:  b'~20014600C06E00010F0CF80CF80CFA0CF90CF70CFA0CF90CF70CF90CFA0CF90CF90CF80CF90CF9050BA50B870B870B870B87FFF6C293A60402C3500001E47C\r'
7e3230303134363030433036453030303130463043463830434638304346413043463930434637304346413043463930434637304346393043464130434639304346393043463830434639304346393035304241353042383730423837304238373042383746464636433239334136303430324333353030303031453437430d
Container:
    NumberOfModule = 1
    NumberOfCells = 15
    CellVoltages = ListContainer:
        3.32
        3.32
        3.322
        3.321
        3.319
        3.322
        3.321
        3.319
[... 132 lines cut ...]
  File "/home/pi/python-pylontech/pylontech/pylontech.py", line 192, in _decode_hw_frame
    assert got_frame_checksum == int(frame_chksum, 16)
AssertionError
```

**Bad:**
```
----> Sending:  b'~20014642E002FFFD0A\r'
7e3230303134363432453030324646464430410d
<---- Receiving:  b''

Traceback (most recent call last):
  File "/home/pi/python-pylontech/tommy_test.py", line 4, in <module>
    print(p.get_values())
  File "/home/pi/python-pylontech/pylontech/pylontech.py", line 284, in get_values
    f = self.read_frame()
  File "/home/pi/python-pylontech/pylontech/pylontech.py", line 214, in read_frame
    f = self._decode_hw_frame(raw_frame=raw_frame)
  File "/home/pi/python-pylontech/pylontech/pylontech.py", line 192, in _decode_hw_frame
    assert got_frame_checksum == int(frame_chksum, 16)
ValueError: invalid literal for int() with base 16: b''

```

Some clue? Perhaps is something normal and the packets in this protocol get corrupted sometimes, more probabbly if they are greater? Does this happen with RS485?

MC68030, 2023-12-29:
Hi tomascrespo,
i had the same idea as you described/ask, let's use the serial console.
Can you please do a fork/share your code ,so we can work together on it? 

I have two UP5000 which i plan to monitor ..
BTW, beside the RS485 communication document, there also exists document from pylontech regarding the RS232 communication.

KR

Frankkkkk, 2023-12-29:
Hi,
I completely forgot about this issue :(. I didn't know that the rs232 proto was documented.; maybe I should find + upload the PDF here?

Indeed, merging both protocols would be awesome! 

Happy new year :-)

MC68030, 2023-12-29:
Just search for "PYLON LFP Battery communication protocol - RS232" :-)