Frankkkkk/python-pylontech #30: AssertionError

Uksa007, 2023-06-04:
Hi,

I get the following error when sending the example data from the PYLON low voltage Protocol RS485 document.

Any ideas what I'm doing wrong?

```
Python 3.11.1 (tags/v3.11.1:a7a450f, Dec  6 2022, 19:58:39) [MSC v.1934 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pylontech
>>> p = pylontech.Pylontech()
>>> print(p.get_values())
b'~20024600F07A11020F0D450D440D450D440D450D440D3E0D450D4A0D4A0D4B0D4A0D4A0D4A0D4A050BC30BC30BC30BCD0BCD0000C725FFFF04FFFF000200CA58012110E1A2\r'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Temp\esphome\pylontech.py", line 271, in get_values
    f = self.read_frame()
        ^^^^^^^^^^^^^^^^^
  File "C:\Temp\esphome\pylontech.py", line 201, in read_frame
    f = self._decode_hw_frame(raw_frame=raw_frame)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Temp\esphome\pylontech.py", line 182, in _decode_hw_frame
    assert got_frame_checksum == int(frame_chksum, 16)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AssertionError

Data sent:
                0x7E, 0x32, 0x30, 0x30, 0x32, 0x34, 0x36, 0x30, // Return battery data
                0x30, 0x46, 0x30, 0x37, 0x41, 0x31, 0x31, 0x30, 0x32, 0x30, 0x46, 0x30, 0x44, 0x34, 0x35, // Battery data
                0x30, 0x44, 0x34, 0x34, 0x30, 0x44, 0x34, 0x35, 0x30, 0x44, 0x34, 0x34, 0x30, 0x44, 0x34,
                0x35, 0x30, 0x44, 0x34, 0x34, 0x30, 0x44, 0x33, 0x45, 0x30, 0x44, 0x34, 0x35, 0x30, 0x44,
                0x34, 0x41, 0x30, 0x44, 0x34, 0x41, 0x30, 0x44, 0x34, 0x42, 0x30, 0x44, 0x34, 0x41, 0x30,
                0x44, 0x34, 0x41, 0x30, 0x44, 0x34, 0x41, 0x30, 0x44, 0x34, 0x41, 0x30, 0x35, 0x30, 0x42,
                0x43, 0x33, 0x30, 0x42, 0x43, 0x33, 0x30, 0x42, 0x43, 0x33, 0x30, 0x42, 0x43, 0x44, 0x30,
                0x42, 0x43, 0x44, 0x30, 0x30, 0x30, 0x30, 0x43, 0x37, 0x32, 0x35, 0x46, 0x46, 0x46, 0x46,
                0x30, 0x34, 0x46, 0x46, 0x46, 0x46, 0x30, 0x30, 0x30, 0x32, 0x30, 0x30, 0x43, 0x41, 0x35, 
                0x38, 0x30, 0x31, 0x32, 0x31, 0x31, 0x30,
                0x45, 0x31, 0x41, 0x32, 0x0D // end of data

```

Uksa007, 2023-06-04:
OK So I calculated the checksum and it seems they go it wrong in the sample response?
Now I get the following errors:
Is the sample response just garbage?

```
>>> print(p.get_values())
b'~20024600F07A11020F0D450D440D450D440D450D440D3E0D450D4A0D4A0D4B0D4A0D4A0D4A0D4A050BC30BC30BC30BCD0BCD0000C725FFFF04FFFF000200CA58012110E27A\r'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Temp\esphome\pylontech.py", line 274, in get_values
    d = self.get_values_fmt.parse(f.info[1:])
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Paul\AppData\Local\Programs\Python\Python311\Lib\site-packages\construct\core.py", line 288, in parse
    return self.parse_stream(io.BytesIO(data), **contextkw)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Paul\AppData\Local\Programs\Python\Python311\Lib\site-packages\construct\core.py", line 300, in parse_stream
    return self._parsereport(stream, context, "(parsing)")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Paul\AppData\Local\Programs\Python\Python311\Lib\site-packages\construct\core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Paul\AppData\Local\Programs\Python\Python311\Lib\site-packages\construct\core.py", line 2120, in _parse
    subobj = sc._parsereport(stream, context, path)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Paul\AppData\Local\Programs\Python\Python311\Lib\site-packages\construct\core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Paul\AppData\Local\Programs\Python\Python311\Lib\site-packages\construct\core.py", line 2653, in _parse
    return self.subcon._parsereport(stream, context, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Paul\AppData\Local\Programs\Python\Python311\Lib\site-packages\construct\core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Paul\AppData\Local\Programs\Python\Python311\Lib\site-packages\construct\core.py", line 2413, in _parse
    e = self.subcon._parsereport(stream, context, path)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Paul\AppData\Local\Programs\Python\Python311\Lib\site-packages\construct\core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Paul\AppData\Local\Programs\Python\Python311\Lib\site-packages\construct\core.py", line 2120, in _parse
    subobj = sc._parsereport(stream, context, path)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Paul\AppData\Local\Programs\Python\Python311\Lib\site-packages\construct\core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Paul\AppData\Local\Programs\Python\Python311\Lib\site-packages\construct\core.py", line 2653, in _parse
    return self.subcon._parsereport(stream, context, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Paul\AppData\Local\Programs\Python\Python311\Lib\site-packages\construct\core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Paul\AppData\Local\Programs\Python\Python311\Lib\site-packages\construct\core.py", line 1041, in _parse
    data = stream_read(stream, self.length, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Paul\AppData\Local\Programs\Python\Python311\Lib\site-packages\construct\core.py", line 91, in stream_read
    raise StreamError("stream read less than specified amount, expected %d, found %d" % (length, len(data)), path=path)
construct.core.StreamError: Error in path (parsing) -> Module -> NumberOfCells
stream read less than specified amount, expected 1, found 0
>>>
```

Frankkkkk, 2023-06-16:
Hi @Uksa007 , I'm sorry but I don't really understand the question; what is your problem ?