Frankkkkk/python-pylontech #35: servere connectivity problems with us5000

ceisserer, 2023-07-18:
I guess this is not an actual bug with the library but with my setup, yet since I didn't find forum to ask related questions, please be kind. I really don't know what else I could try / how to debug this issue further.

When reading the values of my US5000 with an RPI3 it sometimes works (first it failed, with a shorter cable it worked fine for a day and now I am back at the error state) and sometimes I only get:
- RX Exception or
- Error reading serial number
- 1 battery found but after that checksum falures

I already tried:
- reduced speed to 9600 baud
- used two different rs485 adapters (one FTDI based; one based on ch341 but with galvanic insulation)
- used 3 different cables, latest is as short as 5cm. Ethernet cat5e cables cut with brown and brown-white (pin 7 and 8) connected to A/B.
- tried to connect various other PINs to the GND input of the rs485 module, but this never worked even a bit (only RX-Exceptions)

I am really out of options. Could it be some parasitic voltage caused by the switching power supply of the Raspberry?
Any ideas what to try next?

Frankkkkk, 2023-07-18:
hi @ceisserer , can you add some pictures of your setup please ?
Cheers

ceisserer, 2023-07-18:
Hi @Frankkkkk:

Currently the setup is rather simple, just the two PINS (7&8 from the cable) connected to A and B of the RS485 module.
The other cables are un-insulated from former tests, also I tried a 120 Ohm termination resistor between A and B which didn't help:

[image]

With the following Code:
p = pylontech.Pylontech('/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0', 9600)
print(p.get_values())

I get the following output:

[root@localhost tech]# python pylon_nomqtt.py
Traceback (most recent call last):
  File "/home/ce/Privat/Haus/Akku/tech/pylon_nomqtt.py", line 9, in <module>
    print(p.get_values())
          ^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pylontech/pylontech.py", line 270, in get_values
    f = self.read_frame()
        ^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pylontech/pylontech.py", line 200, in read_frame
    f = self._decode_hw_frame(raw_frame=raw_frame)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pylontech/pylontech.py", line 181, in _decode_hw_frame
    assert got_frame_checksum == int(frame_chksum, 16)

So again it seems like communication problems - even at 9600 baud :/
Also it doesn't seem to be caused by parasictic/capacitic currents, I gave it a try with my notebook which was running on battery - same result.

also I have to appologize - I initially used another (more complex) library ('PylontechStack') and the error messages in the first post were from the former library, but I have to admit I like this one more because of its simplicity :)

Frankkkkk, 2023-07-18:
Hi @ceisserer ,
Thanks for the info.
From what I can see on the picture, your DIP1 is off, which (unless I'm mistaken) means that the RS485 baudrate is 115200. Thus:
- Can you try the script with the baudrate unset (or set to 115200, which is the default)
- Can you try to change DIP1 and try again with 9600 ?

If it still doesn't work:
- make sure that the brown/white-brown cables really terminate to pins 7/8 of the RJ45 cable
- try to print the raw_frame (simply edit the pylontech.py file, line 200). Do you always get the same repeatable result ?
- indeed, a 120ohm resistor should be added.

Cheers

ceisserer, 2023-08-14:
The problem was so simple and stupid, it never crossed my mind it could actually be such an easy-to-fix problem.
The source for all the issues observed was a systemd unit file I had created a few weeks before and had forgotten. so when i manually started the script to work on it again after that break one instance was already running and both were reading/writing to the serial device file. i always disliked windows' file locking, but I have to admit in this case it would have saved a lot of time and RJ45 cables I cut open and soldered ;)

Sorry for the noise and thanks a lot for all your help, time and suggestions.

closed 2023-08-14 by ceisserer

Frankkkkk, 2023-08-14:
Hi @ceisserer thanks for the follow up. I've released v0.3.3 which asks for exclusive access to the serial port so that the issue doesn't arise :-)
Cheers

ceisserer, 2023-08-14:
Awesome, thanks a lot!