Frankkkkk/python-pylontech #37: Where do people store data?

phoenix1589, 2023-10-24:
I guess this is more of a discussion than an 'issue', but where / how do people store the data from this for time based graphing?

I'm thinking a every minute cron job to pull the data, and then send it to influxdb. I currently have a single US5000 battery (which has the issue of get_values not working, but get_values_single(2) does).

BoboTiG, 2023-12-03:
Well, that's exactly what I am doing: I push the data to InfluxDb every minutes. On the other side, I used Grafana to display graphs.

Frankkkkk, 2023-12-03:
Same here, but opentsdb + grafana instead of influx. Maybe we could share scripts or dashboards ? :-)

BoboTiG, 2023-12-03:
I do not have yet dashboard, I am in the middle on seting up everything for now. But yes, that would be great to share, and maybe improve them!