Frankkkkk/python-pylontech #39: Help with coding

Minoslo, 2024-02-12:
Is there any chance to help me to implement this code for a GUI?
What I exactly need is to run few commands to get SoC and temperature values. I'm new on this topic and any help would be appreciated. I have been reading the protocol and I think I understand how it works but I don't know how to send a command using a python script.

Frankkkkk, 2024-02-12:
Hi,
A simple "hello" is always welcome :-)
Basically, the library implements most of the commands. There are some examples [here](https://github.com/Frankkkkk/python-pylontech/blob/master/pylontech/pylontech.py#L302).

Once you choose a GUI lib (tkinter for example), you just have to interface with this library and it should work :-)

Maybe I'll write an example if I have the time, but I don't guarantee it.
Cheers

Minoslo, 2024-02-12:
Oh man...I'm so sorry, for real. My comment was rude (english is not my native language so I was stupdly focus trying to explain myself), I apologize. Let me start over.
Hello mate ,
Your lib is really awesome and it had helped me so much on my project. I was able to use most of the commands except "p.get.values()".
I found your project meanwhile I was trying to use the pylon protocol myself but I don't have enough knowledge to do it. I will try to do what you told me. I'm currently using "pysimplegui" to make my GUI. Any example would be appreciated if you have time.
Thank you very much for the fast answer tho.
Cheers.

Frankkkkk, 2024-02-12:
No problem.
Something like that should do the job:
[image]
```python3
import tkinter as tk
from time import time
from pylontech import Pylontech

p = Pylontech()

def update():
    vals = p.get_values()
    bv = vals.Module[0].Voltage
    bi = vals.Module[0].Current
    tot_power = vals.TotalPower
    batv_v.config(text=f"{bv} V")
    bati_v.config(text=f"{bi} A")
    batp_v.config(text=f"{tot_power} W")
    root.after(5000, update)

# Create the main window
root = tk.Tk()
root.title("Pylontech Display")

# Create a label widget for displaying the time
batv_l = tk.Label(root, text="Battery Voltage")
batv_l.grid(column=0, row=0)
batv_v = tk.Label(root, text="xx V")
batv_v.grid(column=1, row=0)

bati_l = tk.Label(root, text="Battery Current")
bati_l.grid(column=0, row=1)
bati_v = tk.Label(root, text="xx A")
bati_v.grid(column=1, row=1)

batp_l = tk.Label(root, text="Battery Power")
batp_l.grid(column=0, row=2)
batp_v = tk.Label(root, text="xx W")
batp_v.grid(column=1, row=2)

time_label2 = tk.Label(root, font=('Helvetica', 48), pady=20)
#time_label2.pack()

# Initialize the time display
update()

# Start the Tkinter event loop
root.mainloop()
```

Cheers

Frankkkkk, 2024-02-12:
PS: If you create a nice UI, it'd be great if you could do an MR someday so we can add it to the repo !

Minoslo, 2024-02-13:
Hello mate 😊,
Thank you a lot! That's the example that I need it to know how the info was grouped within the response. 
When I finish the GUI, you can be sure I will do an MR and if you like the GUI enough, it would be awesome to add it here and help another people. 
Thank you one more time for being so quick to answer.

Cheers

Minoslo, 2025-02-04:
Hello after a long time, mate!

I've been really busy, but I always keep my word. I put together a simple GUI using PySimpleGUI for my batteries, designed to fit a 1024x600 screen. I have 14 modules, so I can select which one I want to view data from a dropdown list (this is, of course, something that each user can customize).
To read the data, I’m using a thread that refreshes everything every 5 seconds.
I’ve never done a merge request before, so I’m using the Issue tab instead. If you like the GUI, just let me know, and I’ll figure out how to add it via MR somehow! 😂

Thanks a lot in advance, and once again, awesome Python package from your side!

Cheers!
Pd: It is in my born language (Spanish) but in case you like it I can just translate it to english.
[image]