Frankkkkk/python-pylontech #40: us5000 reading issue

sante85, 2024-02-20:
When reading the values of my US5000 with an RPI3 not work

I have python 3.11.1

After this commands

'''
Python 3.11.1 (main, Oct 11 2023, 19:09:48) [GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pylontech
>>> p = pylontech.Pylontech()
>>>  print(p.get_values())
'''

this is a output

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/smart-battery-edge/python-pylontech/pylontech/pylontech.py", line 285, in get_values
    f = self.read_frame()
        ^^^^^^^^^^^^^^^^^
  File "/home/smart-battery-edge/python-pylontech/pylontech/pylontech.py", line 215, in read_frame
    f = self._decode_hw_frame(raw_frame=raw_frame)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/smart-battery-edge/python-pylontech/pylontech/pylontech.py", line 196, in _decode_hw_frame
    assert got_frame_checksum == int(frame_chksum, 16)
                                 ^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 16: b''
>>> print(p.get_system_parameters())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/smart-battery-edge/python-pylontech/pylontech/pylontech.py", line 258, in get_system_parameters
    f = self.read_frame()
        ^^^^^^^^^^^^^^^^^
  File "/home/smart-battery-edge/python-pylontech/pylontech/pylontech.py", line 215, in read_frame
    f = self._decode_hw_frame(raw_frame=raw_frame)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/smart-battery-edge/python-pylontech/pylontech/pylontech.py", line 196, in _decode_hw_frame
    assert got_frame_checksum == int(frame_chksum, 16)
                                 ^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 16: b''

Frankkkkk, 2024-02-20:
Hi,
follow the debugging steps in #35  and be more polite in the future :-)

Cheers

closed 2024-02-20 by Frankkkkk

sante85, 2024-02-20:
Response is not clear.

I have a official cable connected at rs232 port.

Which steps I execute to correct run?

best regards

sante85, 2024-02-20:
@Frankkkkk 
can explain me? thanks

sante85, 2024-02-20:
i not understand how to resolve issue

Frankkkkk, 2024-02-20:
Hi,
Please describe your setup, add some pictures, apply the same recommendations as #35.
Cheers

sante85, 2024-02-20:
I have US5000 and i have connected console port with usb of my raspberry pi.

if I do ls /dev/ttyUSB0, one is finded right.

baud by default is 11200 and port by default is /dev/ttyUSB0

now I do these commands
python3
Python 3.11.1 (main, Oct 11 2023, 19:09:48) [GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pylontech
>>> p = pylontech.Pylontech()
>>> print(p.get_values())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/smart-battery-edge/python-pylontech/pylontech/pylontech.py",
line 285, in get_values
    f = self.read_frame()
        ^^^^^^^^^^^^^^^^^
  File "/home/smart-battery-edge/python-pylontech/pylontech/pylontech.py",
line 215, in read_frame
    f = self._decode_hw_frame(raw_frame=raw_frame)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/smart-battery-edge/python-pylontech/pylontech/pylontech.py",
line 196, in _decode_hw_frame
    assert got_frame_checksum == int(frame_chksum, 16)
                                 ^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 16: b''
>>>

How to resolve this issue?

best regards