Frankkkkk/python-pylontech #41: Compatibility with US5000 battery

Benoit3, 2024-03-03:
I belong a bank with a unique U5000C Battery. I use default settings for DIP switch.

get_protocol_version, get_manufacturer_info, get_values are inoperative and return exception.

For example with get_protocol_version:
`
>>> import pylontech
>>> p = pylontech.Pylontech()
>>> print(p.get_protocol_version())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/domotic/python-pylontech/pylontech/pylontech.py", line 242, in get_protocol_version
    return self.read_frame()
  File "/home/domotic/python-pylontech/pylontech/pylontech.py", line 215, in read_frame
    f = self._decode_hw_frame(raw_frame=raw_frame)
  File "/home/domotic/python-pylontech/pylontech/pylontech.py", line 196, in _decode_hw_frame
    assert got_frame_checksum == int(frame_chksum, 16)
ValueError: invalid literal for int() with base 16: b''
>>> 
`

I think issue is coming from the fact that adr=0 is not authorized and should be set to the right value (eg 0x02 in simple/default case case). With this correction, the result is: 
'
>>> import pylontech
>>> p = pylontech.Pylontech()
>>> print(p.get_protocol_version())
Container: 
    ver = b' ' (total 1)
    adr = b'\x02' (total 1)
    cid1 = b'F' (total 1)
    cid2 = b'\x00' (total 1)
    infolength = b'\x00\x00' (total 2)
    info = b'' (total 0)
'
The protocol value seems to be 'space' in Ascii, so 0x20 or V2.0

I need to do same correction for get_manufacturer_info.

For get_values, it's seems to be an other problem, it's seems battery doesn't manage the "system information case"

Benoit3, 2024-03-03:
One question: are usually system information command answered with one bank US2000 battery ?

Frankkkkk, 2024-03-11:
Hi, yeah that would be great :-) Maybe it would be better to edit the README and add a "US2000 vs US5000" section with some code examples ?
Thanks and cheers !